tmolteno/necpp #7: Ruby compile error nec_exception.h: File or Directory not found

marc-hendry, 2015-02-28:
Operating system: Ubuntu 14.04 LTS (64 bit)

After successfully compiling and installing necpp, I tried to compile the Ruby module. However, compilation fails due to missing file "nec_exception.h":

```
$ cd Ruby/ext/necpp
$ ./build.sh

Language subdirectory: ruby
Search paths:
   ./
   ../../../src/
   ./swig_lib/ruby/
   /usr/share/swig2.0/ruby/
   ./swig_lib/
   /usr/share/swig2.0/
Preprocessing...
Starting language-specific parse...
Processing types...
C++ analysis...
Generating wrappers...
checking for clapack.h... yes
checking for clapack_zgetrf() in -llapack... yes
checking for libnecpp.h... yes
checking for nec_create() in -lnecpp... yes
creating Makefile
g++ -I. -I/usr/include/ruby-1.9.1/x86_64-linux -I/usr/include/ruby-1.9.1/ruby/backward -I/usr/include/ruby-1.9.1 -I. -DHAVE_CLAPACK_H -DHAVE_LIBNECPP_H -I/usr/local/include -I/usr/include/atlas -D_FORTIFY_SOURCE=2   -I/home/tim/github/necpp/src -fPIC -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -Wall -fno-strict-aliasing -fPIC -I/home/tim/github/necpp/src  -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -Wall -fno-strict-aliasing -o necpp_wrap.o -c necpp_wrap.cxx
necpp_wrap.cxx:1835:27: fatal error: nec_exception.h: File or Directory not found
 #include "nec_exception.h"
                           ^
compilation terminated.
make: *** [necpp_wrap.o] Error 1
sudo make install
```

Am I missing something?

marc-hendry, 2015-02-28:
OK, found the problem...

In `ext/necpp/extconf.rb` the variable `NECPP_SRC` is set statically. Of course it will not work for other users on other systems in other directories.

Either the user has to manually adjust `NECPP_SRC` or `extconf.rb` needs to be changed in order to automatically detect the path itself.

tmolteno, 2015-03-02:
Thanks for this!

Modifying extconf.rb sounds like the way to go. I'll have a look at it asap.

tmolteno, 2015-03-02:
I've committed a change to the extconf.rb to use a relative path for the nec2++ source directory. Let me know if this works.

Kind Regards

Tim

marc-hendry, 2015-03-04:
Thanks, it works. It compiles right out of the box now.

closed 2015-03-04 by marc-hendry