tmolteno/necpp #8: build problem

nicolasbarbot, 2015-03-03:
Hi,
I try to execute the exemple test_nec.c located in the exemple directory. With the binary version installed with the debian package manager (libneccpp-dev 1.5.0), the build succeeded however the execution failed with:
 gcc -o test_nec test_nec.c -lm -lstdc++ -lnecpp
./test_nec 
Gain: 8.407404, -1.958237 +/- 16.108164 dB
RHCP Gain: 5.397104, -4.960991 +/- 16.072012 dB
LHCP Gain: 5.397104, -4.960991 +/- 16.072012 dB
terminate called after throwing an instance of 'nec_exception*'
Aborted
By building the library form sources (with lapack correctly detected by the configure script in /usr/lib/atlas-base/atlas), i still cannot link the exemple program:
gcc -o test_nec test_nec.c -lm -lstdc++ -L../src/.libs -lnecpp
../src/.libs/libnecpp.so: undefined reference to `clapack_zgetrs'
../src/.libs/libnecpp.so: undefined reference to `clapack_zgetrf'
collect2: error: ld returned 1 exit status
The symbols clapack_zgetrs and clapack_zgetrf are undefined in the libnecpp but are correctly defined in liblaback.
My last question is about the difference of gain values obtained with (for the same antenna) nec2 (or necpp) and the value returned by nec_gain_max() (i observe 2.15 dB for nec2 and 8.36dB for nec_gain_max() for the half wavelength antenna, the others values are equal).

Thank you for your time.

Nicolas Barbot

nicolasbarbot, 2015-03-04:
My compilation problem was due a link with the wrong lapack library:
ldd src/.libs/libnecpp.so | grep lapack
	liblapack.so.3 => /usr/lib/liblapack.so.3 (0x00007fb11078f000)
Rebuilding the library with export LD_LIBRARY_PATH=/usr/lib/atlas-base/atlas solve my issue:
ldd src/.libs/libnecpp.so | grep lapack
	liblapack.so.3 => /usr/lib/atlas-base/atlas/liblapack.so.3 (0x00007f01a1357000)
However the execution of the exemple fail with the same error than the one linked with the library (installed with debian package manager):
gcc -o test_nec test_nec.c -L../src/.libs -lnecpp -lm -lstdc++
./test_nec
Gain: 8.407404, -1.958237 +/- 16.108164 dB
RHCP Gain: 5.397104, -4.960991 +/- 16.072012 dB
LHCP Gain: 5.397104, -4.960991 +/- 16.072012 dB
terminate called after throwing an instance of 'nec_exception*'

Nicolas Barbot

tmolteno, 2015-03-04:
Hi Nicolas

I have updated the example code - it was suffering from Bitrot. The C++ code version displays the exception message. It is...

GEOMETRY DATA ERROR -- FIRST SEGMENT MIDPOINT OF WIRE #2 (TAG ID #0) INTERSECTS WIRE #1 (TAG ID #0)

I'll look into the geometry description, but it looks like it contains intersecting wires. Catching this error was done in a more recent version of nec2++.

I'll commit in a few minutes.

Tim

tmolteno, 2015-03-04:
Hi Nicolas

I have updated the C-style API to include error messages. This has only been applied to a couple of the functions at the moment, but I've updated the examples again to use them. This makes the C API far more useful.

I'll update the remaining API functions in libnecpp.h once I find a little more time :)

Kind Regards

Tim

tmolteno, 2015-03-05:
I have updated all of the void c-style API functions to return int, and to set the 'last-error' message with the contents of the exception thrown. Closing this issue and will open another on the gain differences.

tmolteno, 2015-03-05:
Could you attach a card deck (or source code) for the antenna that gives you the gain difference?

closed 2015-06-08 by tmolteno