tmolteno/necpp #9: Difference in gain values.

tmolteno, 2015-03-05:
Reported as part of a build-related issue. My last question is about the difference of gain values obtained with (for the same antenna) nec2 (or necpp) and the value returned by nec_gain_max() (i observe 2.15 dB for nec2 and 8.36dB for nec_gain_max() for the half wavelength antenna, the others values are equal).

nicolasbarbot, 2015-03-05:
I just see that the difference in gain values were due to a bad GN card used in my C programm. After correction the gain values are exactly the same. Everything is now fine.

Thank you very much for your help and your time.

Nicolas

tmolteno, 2015-03-05:
Great news. Thanks you very much for reporting the problems.

closed 2015-03-05 by tmolteno