tmolteno/necpp #11: Typo in nec_context.cpp

ytakeyasu, 2015-03-07:
Hi,

nec_context.cpp
```
2959 cw[jx*ipr*nw] += etk* m_geometry->ax[ij]+ ets* m_geometry->bx[ij]+ etc* m_geometry->cx[ij];
```
The "jx*" might be "jx+".
```
2959 cw[jx+ipr*nw] += etk* m_geometry->ax[ij]+ ets* m_geometry->bx[ij]+ etc* m_geometry->cx[ij];
```
Regards.

Yoshi Takeyasu

ytakeyasu, 2015-03-07:
Hi again,

I'm wondering also if "nw" is "nr" or not.
```
2959 cw[jx+ipr*nw] += etk* m_geometry->ax[ij]+ ets* m_geometry->bx[ij]+ etc* m_geometry->cx[ij];
```
or
```
2959 cw[jx+ipr*nr] += etk* m_geometry->ax[ij]+ ets* m_geometry->bx[ij]+ etc* m_geometry->cx[ij];
```

Thank you.

Yoshi Takeyasu

tmolteno, 2015-03-07:
Hi Yoshi

Thank you for pointing these out. I will have a look and get back to you.

Kind Regards

Tim Molteno

tmolteno, 2015-03-08:
You are correct with this one. I have modified it to jx+ipr*nr

I will create a testbench to verify any changes to the output.

Kind Regards

Tim

tmolteno, 2015-03-25:
Testbench  completed.

closed 2015-03-25 by tmolteno