tmolteno/necpp #12: for-loop iteration in nec_context.cpp

ytakeyasu, 2015-03-07:
Hi,

nec_context.cpp
```
2466    for( kk = 1; kk < nop; kk++ )
2467			{
2468					deter += scm[kk] * symmetry_array[k+kk*nop];
2469			}
2470			in_cm[row_offset + ka] = deter;
```
Comparing this for-loop with NEC2C's matrix.c :
```
for( kk = 1; kk < nop; kk++ )
{
deter += scm[kk]* ssx[k+kk*nop];
cm[ka+i*nrow]= deter;
}
```
The for-loop might be:
```
2466    for( kk = 1; kk < nop; kk++ )
2467			{
2468					deter += scm[kk] * symmetry_array[k+kk*nop];
2469					in_cm[row_offset + ka] = deter;
2470			}
```

Kind regards.

Yoshi Takeyasu

tmolteno, 2015-03-07:
Hi Yoshi

I think you are correct! I will confirm this as soon as possible and modify the code. I will also try and produce a unit-test that compares the output.

Tim

tmolteno, 2015-03-08:
Hi Yoshi

I have looked more closely at the code, and I think that the index into the array (row_offset + ka)
does not change during the innermost loop. Therefore modifying in_cm should happen outside the loop (so the nec2c code is doing unnecessary operations).

      for( int k = 1; k < nop; k++ ) {
        int ka = j+ k*npeq;
        nec_complex deter = scm[0];

        for( int kk = 1; kk < nop; kk++ ) {
          deter += scm[kk] * symmetry_array[k+kk*nop];
        }
        in_cm[row_offset + ka] = deter; // if this were inside the loop, the result wouldn't change.
      }

Kind Regards

Tim

ytakeyasu, 2015-03-09:
Hi Tim,

Thanks for your review.

Best regards.

Yoshi Takeyasu

closed 2015-03-09 by ytakeyasu