tmolteno/necpp #22: Ruby-Module - Wrong path in swig file

marc-hendry, 2015-05-03:
Using the **build.sh** script works fine, but trying to compile the Ruby module using swig will fail. Seems to be a faulty path to `src/libnecpp.h`

```
marc@ubuntu-12.04-LTS:~/necpp$ swig -v -c++ -ruby necpp.i

Language subdirectory: ruby
Search paths:
   ./
   ./swig_lib/ruby/
   /usr/share/swig2.0/ruby/
   ./swig_lib/
   /usr/share/swig2.0/
Preprocessing...
necpp.i:49: Error: Unable to find '../src/libnecpp.h'
```

tmolteno, 2015-05-03:
Thank you for the report. I have made the path specification consistent (it was not well specified). Now the path to libnecpp.h has to be specified on the command line when using swig. Can you let me know if this works?

Kind Regards

Tim

closed 2015-05-03 by tmolteno

marc-hendry, 2015-05-04:
Thanks, it works fine now.

commit 42c4f362 (2015-05-03): Fixed up paths to be consistent between the README and the build.sh and necpp.i files. The path to libnecpp.h now has to be explicitly specified on the swig command line. Closes #22