tmolteno/necpp #23: Build fails on archlinux and g++ 5.1

vguberinic, 2015-05-22:
Build fails in catch.hpp. Seems to be some issue with nullptr_t being added in c++11. Adding -std=c++11 to src/Makefile seems to solve the issue.

Errors from g++ inlned:
```
In file included from safe_array_tb.cpp:20:0:
catch.hpp:812:5: error: identifier ‘nullptr’ is a keyword in C++11 [-Werror=c++0x-compat]
     inline std::nullptr_t opCast(std::nullptr_t) { return nullptr; }
     ^
In file included from safe_array_tb.cpp:20:0:
catch.hpp:812:17: error: ‘nullptr_t’ in namespace ‘std’ does not name a type
     inline std::nullptr_t opCast(std::nullptr_t) { return nullptr; }
                 ^
catch.hpp:933:58: error: ‘template<Catch::Internal::Operator Op, class T> bool Catch::Internal::compare’ conflicts with a previous declaration
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                          ^
catch.hpp:927:44: note: previous declaration ‘namespace Catch::Internal { }::compare’
     template<Operator Op, typename T> bool compare( T* lhs, int rhs ) {
                                            ^
catch.hpp:933:53: error: ‘nullptr_t’ is not a member of ‘std’
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                     ^
catch.hpp:933:70: error: expected primary-expression before ‘*’ token
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                                      ^
catch.hpp:933:72: error: ‘rhs’ was not declared in this scope
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                                        ^
catch.hpp:933:76: error: expression list treated as compound expression in initializer [-fpermissive]
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                                            ^
catch.hpp:933:44: error: variable templates only available with -std=c++14 or -std=gnu++14 [-Werror]
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                            ^
catch.hpp:933:78: error: expected ‘;’ before ‘{’ token
     template<Operator Op, typename T> bool compare( std::nullptr_t, T* rhs ) {
                                                                              ^
catch.hpp:936:66: error: ‘std::nullptr_t’ has not been declared
     template<Operator Op, typename T> bool compare( T* lhs, std::nullptr_t ) {
                                                                  ^
catch.hpp:936:44: error: redefinition of ‘template<Catch::Internal::Operator Op, class T> bool Catch::Internal::compare(T*, int)’
     template<Operator Op, typename T> bool compare( T* lhs, std::nullptr_t ) {
                                            ^
catch.hpp:927:44: note: ‘template<Catch::Internal::Operator Op, class T> bool Catch::Internal::compare(T*, int)’ previously declared here
     template<Operator Op, typename T> bool compare( T* lhs, int rhs ) {
                                            ^
In file included from safe_array_tb.cpp:20:0:
catch.hpp:1183:38: error: ‘std::string Catch::toString’ redeclared as different kind of symbol
 std::string toString( std::nullptr_t );
                                      ^
catch.hpp:1180:13: note: previous declaration ‘std::string Catch::toString(unsigned char)’
 std::string toString( unsigned char value );
             ^
catch.hpp:1183:23: error: ‘nullptr_t’ is not a member of ‘std’
 std::string toString( std::nullptr_t );
                       ^
In file included from safe_array_tb.cpp:20:0:
catch.hpp:6977:38: error: ‘std::string Catch::toString’ redeclared as different kind of symbol
 std::string toString( std::nullptr_t ) {
                                      ^
catch.hpp:6972:13: note: previous declaration ‘std::string Catch::toString(unsigned char)’
 std::string toString( unsigned char value ) {
             ^
catch.hpp:6977:23: error: ‘nullptr_t’ is not a member of ‘std’
 std::string toString( std::nullptr_t ) {
                       ^
cc1plus: all warnings being treated as errors
Makefile:950: recipe for target 'test_manager-safe_array_tb.o
```

tmolteno, 2015-05-23:
Thanks for this. I added this to the src/Makefile.am. Could  you please check that it works for you.

vguberinic, 2015-05-24:
Works for me.

Won't this however break compatibility with older compilers who don't support c++11?

closed 2015-05-24 by vguberinic