tmolteno/necpp #31: git repo seems inconsistent

jeffsf, 2016-03-16:
While trying to create a patch for #30 I found that a freshly cloned repo is not self-consistent:

`$ git clone https://github.com/tmolteno/necpp
Cloning into 'necpp'...
remote: Counting objects: 2801, done.
remote: Total 2801 (delta 0), reused 0 (delta 0), pack-reused 2801
Receiving objects: 100% (2801/2801), 2.51 MiB | 2.10 MiB/s, done.
Resolving deltas: 100% (2110/2110), done.
Checking connectivity... done.
$ cd necpp/
$ git diff --name-status
M       Python/.gitignore
M       Python/README.md
`
Forcing a hard reset does not resolve
`$ git reset --hard HEAD
HEAD is now at ebfb4d4 Changed API of c_geometry::helix() to put the tag id and segment count at the start.
$ git diff --name-status
M       python/.gitignore
M       python/README.md
`

ctdk, 2017-03-24:
It's because there's `Python/` and `python/` in the repository. Since HFS+ is case preserving, but not case sensitive, it's causing git weirdness.

jeffsf, 2017-03-28:
Ah, thanks @ctdk

closed 2017-03-28 by jeffsf

tmolteno, 2017-08-21:
As the new Python packages reside in a separate python-necpp repository, I will probably merge these into a single 'python' directory with a README pointing to the new repo.