tmolteno/necpp #32: clang build does not permit use of abs() with a float-like argument

jeffsf, 2016-03-16:
```
c_evlcom_tb.cpp:14:15: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
     REQUIRE((abs(j0.real() -  j0r) < eps));
              ^
```

```
$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.3.0
Thread model: posix
```

Easily resolved by using std::fabs()

preich, 2016-06-30:
I had to do this also with gcc-6.1.1 (Fedora 24) to make the software compile.

tmolteno, 2017-08-24:
This has been fixed by recent commits. Thanks for the reports.

closed 2017-08-24 by tmolteno