tmolteno/necpp #35: NEC constantly crashing on connect_segments

kaklik, 2016-05-15:
I am using PyNEC python wrapper and my python code usually crashes on "Segmentation fault (core dumped)" gdb shows following: 

kaklik@UST-vyvoj:~/git/NEC-models/antennas/src$ gdb /usr/bin/python2.7 core
GNU gdb (Ubuntu 7.11-0ubuntu1) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python2.7...(no debugging symbols found)...done.

warning: core file may not match specified executable file.
[New LWP 7490]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `python helix_test.py'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  c_geometry::connect_segments (this=0x1f3b560, ignd=<optimized out>)
    at ../necpp_src/src/c_geometry.cpp:1750
1750	            xa= xa+ x2[ixx];
(gdb)

tmolteno, 2016-06-03:
Hi There, 

Could you send a piece of code that reproduces this?.  It may now be fixed from issue #36.

Kind regards 
Tim

kaklik, 2016-06-07:
I have a quite complex piece of code for now. https://github.com/bolidozor/NEC-models/blob/optimize/antennas/src/SC-QHA.py#L221
But exactly the problem appears if I try to call 
nec.radiation_pattern(thetas=Range(0,0, count=90), phis=Range(0,0,count=90)) 
more times than only once in the program.

tmolteno, 2026-06-26:
Fixed in commit 3155756. The crash in connect_segments was caused by a bounds check using PCHCON (100000) instead of the actual array size n_segments. An invalid segment index between n_segments and PCHCON would pass the check but cause an out-of-bounds array access.\n\nFixes:\n- Changed the outer-loop connection check from `ix <= PCHCON` to `ix <= n_segments`\n- Added explicit bounds checks on `ixx = ix - 1` before every array access (x, x2, y, y2, z, z2, icon1, icon2)\n- Converted silent segfaults into descriptive nec_exception throws with the bad index value\n\nVerified with herzian_dipole.nec — simulation output unchanged.