tmolteno/necpp #36: Helix geometry ignores ITAG value

kaklik, 2016-05-19:
Hello, 

I have designed an antenna model with helical wires, but GH card input seems to ignore ITAG number of helix. Therefore I am unable to feed helix directly by excitation source. 

Here is the NEC code, the output of nec2++ follows.

CM  Helix and screen reflector for 143.05 MHz
CE End Comments
GH     1    50  1.29700E+00  6.48500E-01  1.62100E-01  1.62100E-01  1.62100E-01  1.62100E-01  2.00000E-02
GM     1     3  0.00000E+00  0.00000E+00  9.00000E+01  0.00000E+00  0.00000E+00  0.00000E+00  1.00000E+00
GW    10     2  0.00000E+00  0.00000E+00  6.48500E-01 -1.62100E-01  0.00000E+00  6.48500E-01  1.00000E-02
GW    11     2  0.00000E+00  0.00000E+00  6.48500E-01  1.62100E-01  0.00000E+00  6.48500E-01  1.00000E-02
GW    12     2  0.00000E+00  0.00000E+00  6.48500E-01  0.00000E+00 -1.62100E-01  6.48500E-01  1.00000E-02
GW    13     2  0.00000E+00  0.00000E+00  6.48500E-01  0.00000E+00  1.62100E-01  6.48500E-01  1.00000E-02
SP     0     1 -1.00000E+00 -1.00000E+00 -0.1  1.00000E+00 -1.00000E+00 -0.1  0.00000E+00
SC     0     0  1.00000E+00  1.00000E+00 -0.1  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
GE     0     0  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     1     1      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     2     1      0  0.00000E+00  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     3     1      0 -1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     4     1      0  0.00000E+00 -1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
FR     0     5     0      0  1.40000E+02  1.00000E+01  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
RP     0    25    50      0  0.00000E+00  0.00000E+00  5.00000E+00  5.00000E+00  0.00000E+00  0.00000E+00
EN     0     0     0      0  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00

nec2++  -i QHA_Bolidozor_patch.nec  QHA_Bolidozor_patch.out 
                               NEC++ Runtime Error: 
                               NO SEGMENT HAS AN ITAG OF 1

ytakeyasu, 2016-05-29:
Hi,

This issue is related to:
`Version 1.7.1`
`Changed API of c_geometry::helix() to put the tag id and segment count at the start.`

**c_geometry.cpp:**
`450  helix( xw1, yw1, zw1, xw2, yw2, zw2, rad, card_int_2, card_int_1);`
may be
`450  helix( card_int_1, card_int_2, xw1, yw1, zw1, xw2, yw2, zw2, rad);`

also
**nec_context.cpp:**
`355 m_geometry->helix(s, hl, a1, b1, a2, b2, rad, segment_count, tag_id);`
may be
`355 m_geometry->helix(tag_id, segment_count, s, hl, a1, b1, a2, b2, rad);`

By the way, it seems your model is violating NEC2's modeling rule "wire to be attached to a surface at patch center". To follow this basic rule, you may need multiple smaller surface-patches instead of single large patch. Take a look at the Example 6 of NEC-2 Manual, Part III: User's Guide.

Regards.

Yoshi Takeyasu

tmolteno, 2016-06-03:
Thanks for the feedback. I've modified the helix code and will commit shortly after a little testing.

Kind Regards
Tim

commit b3e67ad8 (2016-06-03): Fixed bug in calling of helix code introduced in last version. Closes #36
closed 2016-06-03 by tmolteno