tmolteno/necpp #37: I can't install python-necpp

lncgomz, 2016-09-22:
Greetings, I would like to use Python-necpp in order to simulate antennas in Python. I downloaded necpp-master folder and followed the instructions for Windows listed [here](https://github.com/tmolteno/python-necpp/blob/master/README.md) when I run _pip install necpp_ in command prompt I get an error message and I can't understand which is exactly the issue. I use Windows 10 and Python 2.7.

We error message goes as follow:

```
Microsoft Windows [Version 10.0.10240]
(c) 2015 Microsoft Corporation. All rights reserved.

C:\Users\MAGO_PC>pip install necpp
Collecting necpp
  Using cached necpp-1.7.0.3.tar.gz
Installing collected packages: necpp
  Running setup.py install for necpp ... error
    Complete output from command C:\Python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\mago_pc\\appdata\\local\\temp\\pip-build-apsxvx\\necpp\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record c:\users\mago_pc\appdata\local\temp\pip-md12hm-record\install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build\lib.win32-2.7
    copying necpp.py -> build\lib.win32-2.7
    running build_ext
    building '_necpp' extension
    creating build\temp.win32-2.7
    creating build\temp.win32-2.7\Release
    creating build\temp.win32-2.7\Release\necpp_src
    creating build\temp.win32-2.7\Release\necpp_src\src
    C:\Users\MAGO_PC\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -DBUILD_PYTHON=1 -Inecpp_src/src/ -Inecpp_src/ -IC:\Python27\include -IC:\Python27\PC /Tpnecpp_src/src\atlas_check.cpp /Fobuild\temp.win32-2.7\Release\necpp_src/src\atlas_check.obj
    atlas_check.cpp
    C:\Users\MAGO_PC\AppData\Local\Programs\Common\Microsoft\Visual C++ for Python\9.0\VC\Bin\cl.exe /c /nologo /Ox /MD /W3 /GS- /DNDEBUG -DBUILD_PYTHON=1 -Inecpp_src/src/ -Inecpp_src/ -IC:\Python27\include -IC:\Python27\PC /Tpnecpp_src/src\c_evlcom.cpp /Fobuild\temp.win32-2.7\Release\necpp_src/src\c_evlcom.obj
    c_evlcom.cpp
    c:\users\mago_pc\appdata\local\temp\pip-build-apsxvx\necpp\necpp_src\src\typesafe_stdint.h(58) : fatal error C1083: Cannot open include file: 'stdint.h': No such file or directory
    error: command 'C:\\Users\\MAGO_PC\\AppData\\Local\\Programs\\Common\\Microsoft\\Visual C++ for Python\\9.0\\VC\\Bin\\cl.exe' failed with exit status 2

    ----------------------------------------
Command "C:\Python27\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\mago_pc\\appdata\\local\\temp\\pip-build-apsxvx\\necpp\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record c:\users\mago_pc\appdata\local\temp\pip-md12hm-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\mago_pc\appdata\local\temp\pip-build-apsxvx\necpp\

```
I need your help in order to resolve this.

rounaksingh, 2016-10-03:
Hi Incgomz,
I am unable to recreate your error. Which C++ build tool u are using?

I used the visual c++ 14 build tools, python2.7, pip 8.1.2, windows 7 to successfully install the necpp.
C:\> pip install necpp
pip installed necpp 1.7.0.3.

Downloaded [Visual C++ build tools](http://landinghub.visualstudio.com/visual-cpp-build-tools). give it a try.

rounaksingh, 2016-10-03:
Hi,
I am getting similar warnings. If I do pip install in windows. I think the pip is not provided with the options to compile necpp for windows. That's why you are getting errors.

rounaksingh, 2016-10-03:
Why don't you try python 3 instead of python2.7

I tried it's working. Just install python3.5,  [(visualc++ build tools 14)](landinghub.visualstudio.com/visual-cpp-build-tools) and do 

pip install necpp

rounaksingh, 2016-10-03:
Try below code

`import necpp

def handle_nec(result):
  if (result != 0):
    print(necpp.nec_error_message())

def impedance(frequency, z0, height):

  nec = necpp.nec_create()
  handle_nec(necpp.nec_wire(nec, 1, 17, 0, 0, z0, 0, 0, z0+height, 0.1, 1, 1))
  handle_nec(necpp.nec_geometry_complete(nec, 1))
  handle_nec(necpp.nec_gn_card(nec, 1, 0, 0, 0, 0, 0, 0, 0))
  handle_nec(necpp.nec_fr_card(nec, 0, 1, frequency, 0))
  handle_nec(necpp.nec_ex_card(nec, 0, 0, 1, 0, 1.0, 0, 0, 0, 0, 0)) 
  handle_nec(necpp.nec_rp_card(nec, 0, 90, 1, 0,5,0,0, 0, 90, 1, 0, 0, 0))
  result_index = 0

  z = complex(necpp.nec_impedance_real(nec,result_index), 
              necpp.nec_impedance_imag(nec,result_index))

  necpp.nec_delete(nec)
  return z

if __name__ == '__main__':
  print("Starting ...")
  z = impedance(frequency = 34.5, z0 = 0.5, height = 4.0)
  print("Impedance \t({},{}I) Ohms".format(z.real, z.imag))`

lncgomz, 2016-10-04:
I resolved this issue by installing Python 3 (VC++ Build Tools 14) and, running pip install necpp. Thanks for your help.

tmolteno, 2017-08-22:
I will close this issue as it appears to be working now. Please reopen if necessary.

closed 2017-08-22 by tmolteno

shinong, 2018-04-17:
Hi guys, 
I am still facing this issue. I am using python 3.6. And I was trying to install the lib by pip, but I still have the Visual C++ issue. I have already installed the building tool. But the installation just did not go through. Can anyone help me to fix this issue?
Thanks

tmolteno, 2018-04-27:
Hi There, 

I don't have easy access to any Windows based computers, but it may be possible to bypass the building step by creating a binary distribution. I will try to borrow one and report back

Kind Regards
Tim