tmolteno/necpp #38: GUI Implementation

lncgomz, 2016-10-04:
Greetings,
I am a telecommunications engineer college student from University of Carabobo (Venezuela) and i'm looking for a thesis subject, in that sense , I would like to implement a graphical user interface for Python-NECpp code, I've been reading the necpp guide and found out that there are some "to-do list" for adding new features. It would be posible to bind my thesis with Python-necpp development?

guitorri, 2016-10-30:
Why not adopt an existing GUI and extend it? 
Last time (Oct 2014) I checked with the author of [QAntenna](http://qantenna.sourceforge.net/index.php). He is no longe working on it and he is willing to pass it over to someone who could lead the project further.
Just an idea...

tmolteno, 2017-08-28:
Hi There. You would be welcome to do this.

lncgomz, 2017-08-28:
I will work on that as soon as possible! It would be ok if a use java in order to implement this GUI?

tmolteno, 2017-08-28:
Definitely. If you create a github repository for it, I'll link to it from the README.md

lncgomz, 2017-10-09:
A Java-based GUI for necpp is available in https://github.com/lncgomz/UCNEC this is my undergraduate thesis work. I hope this be useful, it's a first version and is written in spanish, I will be working very soon in a english version.

tmolteno, 2017-10-10:
This is excellent. I will put a link to it from the main README on necpp.