tmolteno/necpp #41: output NEC2 cards

rcnlee, 2016-12-03:
I am using the API to create input, i.e., nec_create(), nec_wire(), nec_fr_card(), etc., is there a way to output the context/configuration to file as a traditional nec2 card deck?

Thanks!

swissel, 2017-02-17:
I would also like to know the answer to this question.

lncgomz, 2017-08-18:
I don't understand your question, however, I would like to help. Could you explain what do you mean with context/configuration?

tmolteno, 2017-08-20:
Hi There,

As I see it, the request is to be able to generate nec2 cards from a geometry description in code. At the moment there isn't a way to do this. My current efforts are to create a grammar to describe geometry that is more expressive than the nec2 cards (using a modern parsing toolkit). It might be possible, once this is done, to compile the grammar to nec2 cards -- I will have to think a bit so see if this is a good way to do things.

Kind Regards,
Tim

rcnlee, 2017-08-25:
As a workaround, I created a Julia package that wraps necpp python.  As the user calls the individual functions (like wire), I intercept it, format it, and spit it out to file to get the NEC card output.  You can take a look at it here:  https://github.com/rcnlee/necpp.jl.  It's a quick and dirty solution, but it worked for me for my small experiment.

tmolteno, 2017-08-27:
Thats pretty cool. I will add a links section to the main README that points to it.

jsr38, 2017-08-27:
Have you started on your efforts to create a geometry grammar?  I have some experience (a long while ago with antlr but clearly this is a Java tool) in doing this sort of thing.  I see there is a boost library, Spirit, that looks suitable but I guess it depends on how you feel about that as a dependency.  Also, I would really like to be able to generate calls to a CSG library or generate OpenSCAD code in order to visualise antenna structures.  Should I open a ticket and if I were to work on something should I steer clear of the geometry grammar as you said you're currently placing effort into this?  The two are linked, though, and I haven't thought through if there are knock-on consequences for choices in one vs. the other...

lncgomz, 2017-08-27:
Hi everyone!
I am developing a GUI based on NEC2++ which can generate geometry .nec files from a table in which an user insert wires coordenates, diameter and segments quantity.; in fact, a whole simulation description can be generated in order to execute NEC2++ (From Windows or Linux distribution), obtain an output file and see radiattion pattern (among other parameters) in a graphical way.

tmolteno, 2017-08-27:
There is a directory called /antlr in the source distribution. This has been on my todo list for a while, and I've just checked in a minor bitrot fix. The grammar file is nec.g.

tmolteno, 2026-08-17:
Closing as stale — open since 2016 with no action. There is still no card-deck serializer: nec_parse reads decks into a context, but nothing writes a context back out as NEC-2 cards. If this resurfaces, reopen with the current API usage and desired output format.

closed 2026-08-17 by tmolteno