tmolteno/necpp #42: Undefined references in examples

KajWiik, 2017-02-11:
This is in Ubuntu Xenial 16.04
``
$ make test_c
gcc -o test_c -Wl,-rpath -Wl, test_nec.o -lnecpp
//usr/local/lib/libnecpp.so: undefined reference to `clapack_zgetrs'
//usr/local/lib/libnecpp.so: undefined reference to `clapack_zgetrf'
collect2: error: ld returned 1 exit status
Makefile:34: recipe for target 'test_c' failed
make: *** [test_c] Error 1
``
``$ make test_cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o test_cpp.o test_cpp.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/c_ggrid.o ../src/c_ggrid.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/matrix_algebra.o ../src/matrix_algebra.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_ground.o ../src/nec_ground.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/c_evlcom.o ../src/c_evlcom.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/c_plot_card.o ../src/c_plot_card.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/misc.o ../src/misc.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_context.o ../src/nec_context.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_output.o ../src/nec_output.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/c_geometry.o ../src/c_geometry.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_exception.o ../src/nec_exception.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_radiation_pattern.o ../src/nec_radiation_pattern.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_structure_currents.o ../src/nec_structure_currents.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/electromag.o ../src/electromag.cpp
g++ -I ../src/ -I .. -O2 -Wall   -c -o ../src/nec_results.o ../src/nec_results.cpp
gcc -o test_cpp test_cpp.o ../src/c_ggrid.o ../src/matrix_algebra.o ../src/nec_ground.o ../src/c_evlcom.o ../src/c_plot_card.o ../src/misc.o ../src/nec_context.o ../src/nec_output.o ../src/c_geometry.o ../src/nec_exception.o ../src/nec_radiation_pattern.o ../src/nec_structure_currents.o ../src/electromag.o ../src/nec_results.o -lstdc++ -lm -llapack
./test_cpp
Nec2++ C++ example. Running (takes a few minutes...)
./test_cpp: symbol lookup error: ./test_cpp: undefined symbol: clapack_zgetrf
Makefile:62: recipe for target 'test_cpp' failed
make: *** [test_cpp] Error 127
``

rounaksingh, 2017-03-03:
Before running that, try installing libatlas-base-dev libatlas-headers as mentioned in [https://bugs.launchpad.net/ubuntu/+source/necpp/+bug/511207] and [https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=531533]

tmolteno, 2017-08-21:
I have updated the README to reflect this in the examples directory. Also fixed a bug in the C test code that was using nec_geometry_complete with only one of the integer parameters. The API now requires both.

closed 2017-08-21 by tmolteno