tmolteno/necpp #43: Frequency loop don't work with plane wave excitation

jpadam, 2017-02-16:
In nec_context.cpp
excitation_loop never return FREQ_LOOP_CONTINUE, only FREQ_PRINT_NORMALIZATION if m_excitation_type is not 0 or 4 (which is the case of the plane wave)

tmolteno, 2017-08-21:
Will look into this. Can you send a small example piece of code, with the expected output.

tmolteno, 2017-08-28:
Issue confirmed as a bug. I have a test case in ./testharness/data/plane_wave_excitation.nec.

jpadam, 2017-09-06:
Sorry for my late answer. I confirm that the test case you added reproduce the bug I observed.

StefanBruens, 2023-02-19:
`nfrq`  is only checked for `VOLTAGE`, `CURRENT` and `VOLTAGE_DISC` excitation.

https://github.com/tmolteno/necpp/blob/870f110875ef7e1fc951cbd489fde20ac9a93284/src/nec_context.cpp#L1754-L1768

Oh, and test_manager now crashes when trying to dereference `rp`, which is a `nullptr` 
https://github.com/tmolteno/necpp/blob/870f110875ef7e1fc951cbd489fde20ac9a93284/src/nec_context_tb.cpp#L140-L141