tmolteno/necpp #45: Build fails on safe array 'for' clause (-Werror=misleading-indentation)

Pbtflakes, 2017-03-08:
Configuration was done with Lapack disabled, Makefile was invoked using `make -j 4`, this is the first error encountered.

Build log:
```
In file included from safe_array_tb.cpp:20:0:
catch.hpp: In member function ‘bool Catch::TestSpec::Filter::matches(const Catch::TestCaseInfo&) const’:
catch.hpp:2813:17: error: this ‘for’ clause does not guard... [-Werror=misleading-indentation]
                 for( std::vector<Ptr<Pattern> >::const_iterator it = m_patterns.begin(), itEnd
= m_patterns.end(); it != itEnd; ++it )
                 ^~~
catch.hpp:2816:21: note: ...this statement, but the latter is misleadingly indented as if it is
guarded by the ‘for’
                     return true;
                     ^~~~~~
```
Compiler version:
```
g++ (SUSE Linux) 6.3.1 20170202 [gcc-6-branch revision 245119]
```

jdpf, 2017-04-11:
I can confirm that this build issue appears on Debian Stretch also... 

```
g++ -DHAVE_CONFIG_H -I. -I..    -std=c++11 -O0 -g3 -Wall -Wextra -Werror -DNEC_ERROR_CHECK=1 -g -O2 -MT test_manager-safe_array_tb.o -MD -MP -MF .deps/test_manager-safe_array_tb.Tpo -c -o test_manager-safe_array_tb.o `test -f 'safe_array_tb.cpp' || echo './'`safe_array_tb.cpp
In file included from safe_array_tb.cpp:20:0:
catch.hpp: In member function ‘bool Catch::TestSpec::Filter::matches(const Catch::TestCaseInfo&) const’:
catch.hpp:2813:17: error: this ‘for’ clause does not guard... [-Werror=misleading-indentation]
                 for( std::vector<Ptr<Pattern> >::const_iterator it = m_patterns.begin(), itEnd = m_patterns.end(); it != itEnd; ++it )
                 ^~~
catch.hpp:2816:21: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
                     return true;
                     ^~~~~~
cc1plus: all warnings being treated as errors
```

With the following compiler: g++ (Debian 6.2.1-5) 6.2.1 20161124

I note that the failure is in the test_manager section, and that removing -Werror from the Makefiles temporarily, or overriding the CXXFLAGS environment variable does not enable a working build. I get the following output on Debian Stretch:

```
g++ -DHAVE_CONFIG_H -I. -I..    -std=c++11 -O0 -g3 -Wall -Wextra -DNEC_ERROR_CHECK=1 -g -O2 -MT test_manager-c_geometry_tb.o -MD -MP -MF .deps/test_manager-c_geometry_tb.Tpo -c -o test_manager-c_geometry_tb.o `test -f 'c_geometry_tb.cpp' || echo './'`c_geometry_tb.cpp
In file included from safe_array_tb.cpp:20:0:
catch.hpp: In member function ‘bool Catch::TestSpec::Filter::matches(const Catch::TestCaseInfo&) const’:
catch.hpp:2813:17: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
                 for( std::vector<Ptr<Pattern> >::const_iterator it = m_patterns.begin(), itEnd = m_patterns.end(); it != itEnd; ++it )
                 ^~~
catch.hpp:2816:21: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
                     return true;
                     ^~~~~~
mv -f .deps/test_manager-nec_results.Tpo .deps/test_manager-nec_results.Po
g++ -DHAVE_CONFIG_H -I. -I..    -std=c++11 -O0 -g3 -Wall -Wextra -DNEC_ERROR_CHECK=1 -g -O2 -MT test_manager-c_evlcom_tb.o -MD -MP -MF .deps/test_manager-c_evlcom_tb.Tpo -c -o test_manager-c_evlcom_tb.o `test -f 'c_evlcom_tb.cpp' || echo './'`c_evlcom_tb.cpp
mv -f .deps/test_manager-c_geometry_tb.Tpo .deps/test_manager-c_geometry_tb.Po
/bin/bash ../libtool  --tag=CXX   --mode=link g++  -g -O2  -L/usr/lib/atlas-base/atlas -o nec2diff necDiff.o  -lm 
In file included from c_evlcom_tb.cpp:1:0:
c_evlcom_tb.cpp: In function ‘void test_bessel(nec_float, nec_float, nec_float, nec_float)’:
c_evlcom_tb.cpp:14:35: error: call of overloaded ‘abs(double)’ is ambiguous
      REQUIRE((abs(j0.real() -  j0r) < eps));
                                   ^
catch.hpp:1478:31: note: in definition of macro ‘INTERNAL_CATCH_TEST’
             ( __catchResult->*expr ).endExpression(); \
                               ^~~~
c_evlcom_tb.cpp:14:6: note: in expansion of macro ‘REQUIRE’
      REQUIRE((abs(j0.real() -  j0r) < eps));
      ^~~~~~~
In file included from /usr/include/c++/6/cstdlib:75:0,
                 from /usr/include/c++/6/ext/string_conversions.h:41,
                 from /usr/include/c++/6/bits/basic_string.h:5402,
                 from /usr/include/c++/6/string:52,
                 from /usr/include/c++/6/bits/locale_classes.h:40,
                 from /usr/include/c++/6/bits/ios_base.h:41,
                 from /usr/include/c++/6/ios:42,
                 from /usr/include/c++/6/istream:38,
                 from /usr/include/c++/6/sstream:38,
                 from catch.hpp:60,
                 from c_evlcom_tb.cpp:1:
```

SNIP other similar errors...

Also, it may be of some use to list the dependency on LAPACK from atlas:

`sudo apt-get install libatlas-base-dev`

Help?

jdpf, 2017-04-13:
I can further confirm that this build process fails on Arch also, so it seems to indeed, be a function of the code here. Again, any hints on how to address this problem would be appreciated. 

Thanks!

rounaksingh, 2017-05-26:
I think this is related to g++ 6 compiler. 

Please read

- https://stackoverflow.com/questions/37482904/why-didnt-the-compiler-warn-me-about-an-empty-if-statement

- https://gcc.gnu.org/gcc-6/porting_to.html

- https://github.com/pornel/pngquant/issues/215

- https://github.com/dut-lang/squirrel/commit/36c308bf57fa5768d2f812a1cdd2cb10357f812f

tmolteno, 2017-08-21:
I am in the process of updating the build files to work with gcc 7.0.

commit 281961eb (2017-08-21): Added catch.hpp most recent version. Closes #45
closed 2017-08-21 by tmolteno