tmolteno/necpp #48: PyNec helix geometry produces no wires

Wanderer43, 2017-09-15:
Hello,

It looks like this is the similar issue to [Helix geometry ignores ITAG value #36](https://github.com/tmolteno/necpp/issues/36) but this time for PyNEC from necpp 1.7.3.2.

When I create a helix I get an error

Traceback (most recent call last):
  File "test_rp.py", line 16, in <module>
    context.geometry_complete(0)
  File "/usr/local/lib/python2.7/dist-packages/PyNEC.py", line 143, in geometry_complete
    return _PyNEC.nec_context_geometry_complete(self, *args)
RuntimeError: Geometry has no wires or patches.

My .py file is below

import matplotlib
matplotlib.use('Agg')

from PyNEC import *
import numpy as np

#creation of a nec context
context=nec_context()

#get the associated geometry
geo = context.get_geometry()

#add wires to the geometry
#geo.wire(0, 36, 0, 0, 0, -0.042, 0.008, 0.017, 0.001, 1.0, 1.0)
geo.helix(0.01,0.1,0.015,0.015,0.015,0.015,0.01,100,0)
context.geometry_complete(0)

context.gn_card(-1, 0, 0, 0, 0, 0, 0, 0)

#add a "ex" card to specify an excitation
context.ex_card(1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0)

#add a "fr" card to specify the frequency 
context.fr_card(0, 2, 2400, 100.0e6)

#add a "rp" card to specify radiation pattern sampling parameters and to cause program execution
context.rp_card(0, 91, 1, 0, 5, 0, 0, 0.0, 45.0, 4.0, 2.0, 1.0, 0.0)

#get the radiation_pattern
rp = context.get_radiation_pattern(0)

#Gains are in decibels
gains_db = rp.get_gain()
gains = 10.0**(gains_db / 10.0)
thetas = rp.get_theta_angles() * 3.1415 / 180.0
phis = rp.get_phi_angles() * 3.1415 / 180.0

#Plot stuff
import matplotlib.pyplot as plt

ax = plt.subplot(111, polar=True)
ax.plot(thetas, gains[:,0], color='r', linewidth=3)
ax.grid(True)

ax.set_title("Gain at an elevation of 45 degrees", va='bottom')
plt.savefig('RadiationPattern.png')
plt.show()

By the way, I have not found a helix card in necpp module.

Thanks

tmolteno, 2017-09-15:
Thanks for the report.  Will look into it this weekend.

GeorgyDemyanov, 2024-04-17:
geo.wire(0, 36, 0, 0, 0, -0.042, 0.008, 0.017, 0.001, 1.0, 1.0)

for what reasons did you create just such a wire?