tmolteno/necpp #50: Get rid of two loops.

livep2000, 2017-11-01:
Hello Tim, excellent work !

I found a possible way of getting rid of 2 loops in 'void solves'. 
At line 705 and 755, this piece of exact equal code:

```
for (int64_t k = 0; k < nop; k++ ) {
        int64_t ia= i+ k* npeq;
        scm[k]= b[ia+column_offset];
      }

      nec_complex sum_normal(scm[0]);
      for (int64_t k = 1; k < nop; k++ )
        sum_normal += scm[k];

	/* test added */
	nec_complex tester = solves_sum_cpl(b, scm, nop, npeq, i, column_offset);	
	ASSERT(tester == sum_normal);

        // ...... more code
}
```
I ASSERT tested the following function, but could not find a case where it is triggered.
So, I commented out '**if ( nop == 1) return**;'  a few lines above.

```
/* I: Test to see if we can snip out a loop             scm = referenced    */
nec_complex solves_sum_cpl(complex_array b, complex_array& scm, int64_t nop, int64_t npeq, int64_t i, int64_t column_offset)
{
	nec_complex sum_normal;
	for (int64_t k = 0; k < nop; k++)
	{
		scm[k] = b[(i + k * npeq) + column_offset];
		sum_normal += scm[k];
	}
	return sum_normal;
}
```
It integrates summing into its parent loop.

Imre Biacsics

tmolteno, 2017-11-02:
Hi Imre,

This looks excellent. I'll double check, but otherwise will incorporate.

livep2000, 2017-11-04:
Hello Tim,

Sorry for this bulky mail, you must have more important things todo then  
reading.
But I have some proposals and need your opinion on that.

(Sorry for the bad grammar, native language is Dutch)

Why was I optimizing  a small piece of an fairley huge project?

Wel. i am a frequent user of Arie's 4nec2, wich can do everything,
but when it comes to modelling the antenna it's not that handy.

I started a project in c#, with the help of the 'helix-toolkit' (3d) to  
create a more modern modeller.
That part was almost ready, when I realized that there was a lack of  
knowledge about the input and output fomat of the files.
And realy? Using files, based on punchcards to interface between software?

I started to translate nec++ to c#! WHAT WAS I THINKING!  At the point of  
debugging I discovered that i had
translated a very old port and that the code since then was maintained and  
improved. (Sigh)

So, most of it is redone. And that was a steep learning curve, because i  
have seen the pretty plain first translation.
And your last branche. Excellent work, you implemented many things, that  
came to my mind while tranlating the first time.

The nec_3vector for example, got a working c# variant, that needs to be  
tested. Got a step further by wrapping the 3 indexed
array again in a 3d points class. (Wich can be recylced on almost every  
part of your code)

Let me be clear: "The chance that my project will succeed is nearly zero".  
Completely stuck in way c++ does variable references.
It is just too large for what is yields.

Downsides :

- c# is known slower dan c++
- Breaks the multiple platform idea.

Benefit:

- Much easier (At last for me) to maintain an improve.
- Direct API's in windows.
- c# lists, collections, library.... well they make live easy.
- Hmmmm, threading!?

So, take a turn and see how we can achieve the same goal, with less  
effort. (stick to c++)

Goal:

1. A why to directly interface with other software, without the use of  
files.
* C++ can do that, on linux based systems by piping or even tcp connection
* Windows also, by creating native .dll

2. Create unified input and output format, that can connect with every  
possible source or destination.
* Of course the traditional .nec and .out files to prevent breaking  
compatibility
* New filecontent based on Json.
(Is that conform with issue #41 , output as nec cards?)

Thus, stick to c++. Then, why not participating in your project? I  
understand a crap of all those calculations.
(Well, not exactly true, after rewriting 1200+ functions TWICE!)
But know how to write lean code, and like to modernize te input and output  
parts of nec++

How does that sound?

Imre Biacsics

livep2000, 2017-11-04:
Oops, reacted by email. Removed it from github. ((to bulky)
The content in shorter terms: 
"I want to exchange some ideas about the nec2 input and output."
Imre