tmolteno/necpp #52: clapack.h not found

mhdry, 2018-02-16:
In Ubuntu 16.04 LTS, with properly intalled `g++`, `make`, `automake`, `autoconf`, `libtool` and `libatlas-base-dev` compilation will fail at:

`matrix_algebra.cpp:412:21: fatal error: clapack.h: File or directory not found`

rounaksingh, 2018-02-18:
Please read the install.md for building instructions. You can install necpp without lapack. or else install "the appropriate atlas system aptitude install libatlas-base-dev"[1].

[1] https://github.com/tmolteno/necpp/blob/master/INSTALL.md

rounaksingh, 2018-02-18:
Hi @mhdry ,
Can you please post the ouput of the configure and build here?

Also please check the location of clapack.h. The .h file should be in /usr/include/atlas.
Therefore, the configure and make should include this location.

In the output it should say like
`ATLAS_INCLUDE_DIR = /usr/include/atlas`
If not then you have found your error. 

Please tell what happens so that it can be fixed.

mhdry, 2018-02-27:
There is 

`-rw-r--r-- 1 root root 9708 Nov 24  2015 /usr/include/atlas/clapack.h`

on my system, but I can't see `ATLAS_INCLUDE_DIR = /usr/include/atlas` in any of the output:

**(1) make -f Makefile.git**
```
libtoolize --force --copy
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
aclocal
autoheader
automake --add-missing
configure.ac:6: installing './compile'
configure.ac:6: installing './config.guess'
configure.ac:6: installing './config.sub'
configure.ac:4: installing './install-sh'
configure.ac:4: installing './missing'
src/Makefile.am: installing './depcomp'
autoconf
```

**(2) ./configure CXX=g++-5 CXXFLAGS="-O2 -march=nocona -pipe"**
```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
[... 80 lines cut ...]
mv -f .deps/libnecpp_la-c_plot_card.Tpo .deps/libnecpp_la-c_plot_card.Plo
/bin/bash ../libtool  --tag=CXX   --mode=compile g++-5 -DHAVE_CONFIG_H -I. -I..    -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-libNEC.lo -MD -MP -MF .deps/libnecpp_la-libNEC.Tpo -c -o libnecpp_la-libNEC.lo `test -f 'libNEC.cpp' || echo './'`libNEC.cpp
libtool: compile:  g++-5 -DHAVE_CONFIG_H -I. -I.. -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-c_ggrid.lo -MD -MP -MF .deps/libnecpp_la-c_ggrid.Tpo -c c_ggrid.cpp -o libnecpp_la-c_ggrid.o >/dev/null 2>&1
libtool: compile:  g++-5 -DHAVE_CONFIG_H -I. -I.. -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-libNEC.lo -MD -MP -MF .deps/libnecpp_la-libNEC.Tpo -c libNEC.cpp  -fPIC -DPIC -o .libs/libnecpp_la-libNEC.o
libtool: compile:  g++-5 -DHAVE_CONFIG_H -I. -I.. -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-c_evlcom.lo -MD -MP -MF .deps/libnecpp_la-c_evlcom.Tpo -c c_evlcom.cpp -o libnecpp_la-c_evlcom.o >/dev/null 2>&1
mv -f .deps/libnecpp_la-c_ggrid.Tpo .deps/libnecpp_la-c_ggrid.Plo
/bin/bash ../libtool  --tag=CXX   --mode=compile g++-5 -DHAVE_CONFIG_H -I. -I..    -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-matrix_algebra.lo -MD -MP -MF .deps/libnecpp_la-matrix_algebra.Tpo -c -o libnecpp_la-matrix_algebra.lo `test -f 'matrix_algebra.cpp' || echo './'`matrix_algebra.cpp
libtool: compile:  g++-5 -DHAVE_CONFIG_H -I. -I.. -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-libNEC.lo -MD -MP -MF .deps/libnecpp_la-libNEC.Tpo -c libNEC.cpp -o libnecpp_la-libNEC.o >/dev/null 2>&1
libtool: compile:  g++-5 -DHAVE_CONFIG_H -I. -I.. -O2 -g0 -Wall -Wextra -Wshadow -Wconversion -O2 -march=nocona -pipe -MT libnecpp_la-matrix_algebra.lo -MD -MP -MF .deps/libnecpp_la-matrix_algebra.Tpo -c matrix_algebra.cpp  -fPIC -DPIC -o .libs/libnecpp_la-matrix_algebra.o
matrix_algebra.cpp:412:21: fatal error: clapack.h: Datei oder Verzeichnis nicht gefunden
compilation terminated.
Makefile:673: die Regel für Ziel „libnecpp_la-matrix_algebra.lo“ scheiterte
make[2]: *** [libnecpp_la-matrix_algebra.lo] Fehler 1
make[2]: *** Auf noch nicht beendete Prozesse wird gewartet …
mv -f .deps/libnecpp_la-c_evlcom.Tpo .deps/libnecpp_la-c_evlcom.Plo
mv -f .deps/libnecpp_la-libNEC.Tpo .deps/libnecpp_la-libNEC.Plo
[... cut ...]  -MT libnecpp_la-c_geometry.lo -MD -MP -MF .deps/libnecpp_la-c_geometry.Tpo -c c_geometry.cpp -o libnecpp_la-c_geometry.o >/dev/null 2>&1
mv -f .deps/libnecpp_la-c_geometry.Tpo .deps/libnecpp_la-c_geometry.Plo
make[2]: Verzeichnis „/home/user/necpp/src“ wird verlassen
Makefile:470: die Regel für Ziel „all-recursive“ scheiterte
make[1]: *** [all-recursive] Fehler 1
make[1]: Verzeichnis „/home/user/necpp“ wird verlassen
Makefile:379: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```

arturrataj, 2018-12-18:
Ubuntu 18.04, no clapack.h at all after installing libatlas-base-dev libatlas3-base. After installing anything containing "libatlas" i.e. apt install libatlas*, clapack.h can be found in /usr/include/x86_64-linux-gnu/clapack.h. Yet, some applications can not find it there.

shiningsurya, 2020-05-30:
Hi, 

I faced the same problem, this is the work around which worked for me:
`CXXFLAGS=-I<path>` and `CFLAGS=-I<path>`
where <path> is the path to the directory which has the `clapack.h` include file. 

I didn't see any configure option which lets you specify the atlas build location.

SimplebutComplex, 2022-03-03:
on ubuntu 20.04  clapack.h is located at 
/usr/lib/x86_64-linux-gnu