tmolteno/necpp #53: RCS computation of dihedral corner reflector

vineethajoy, 2018-03-14:
I was trying to compute the monostatic RCS of a dihedral corner reflector(Plate length=0.3m, Plate thickness=0.005m, f=9.4GHz) using NEC. But somehow I am not getting correct results. The output I am getting is given below and the NEC input file is attached herewith.
[image]

[PEC-Dihedral-Tab-Delimiter-Tag-no-modify.txt](https://github.com/tmolteno/necpp/files/1809477/PEC-Dihedral-Tab-Delimiter-Tag-no-modify.txt)

Any suggestion would be really helpful.

ytakeyasu, 2018-03-17:
As NEC2++ output Runtime Error for your model:
GEOMETRY DATA ERROR -- WIRE #1 (TAG ID #1) INTERSECTS WIRE #4 (TAG ID #124)
I tried the model using 4nec2 & NEC2DXS on Windows and got the attached result. I suppose the result is what you expected. It seems NEC2++ and/or NEC2 is not the problem.

vineethajoy, 2018-03-17:
Thank you for your response. 
I tried the same in FEKO and I am getting similar results there too.  But the correct plot is as given below:
[image]

I think there is some fundamental problem with the modelling. But I am not able to figure it out.

ytakeyasu, 2018-03-17:
I'll suggest you to review your description of "EX" and "RP" command in accordance with NEC2's definition. The issue may be the direction of incident plane wave.

vineethajoy, 2018-03-17:
Exactly. The model I dealt with is the cross-section in the XY plane i.e. Fig. 3 .  
The results from comsol were perfectly matching with the results from an MoM code we wrote by our own and that was experimentally verified too.
 I did try various changes in the EX card like different values for theta, phi and eta. ETA= 0/180 gave no output whereas ETA=90/270 gave similar results. I also tried plotting in various planes (XY,YZ and XZ) by modifying RP card. But  still no luck.

The multiple bounce effect around the corner is not coming into picture.

vineethajoy, 2018-03-23:
As the thickness of the strip is very small compared to wavelength, I think the solvers are ultimately treating them as wires instead of strips. 
I tried a Dihedral two-plates problem (3D problem) and I am getting correct results. The plots are attached herewith. 
Problem dimensions:
Problem Satement:
100cm*100cm
f=600MHz
Segment length=0.04
Segment radius=0.002

Ref: Fig.3.33 (P.no: 204) of the book "Complex electromagnetic problems and numerical simulation approaches" by Levent Sevgi.
[image]
[image]

closed 2018-03-23 by vineethajoy

ytakeyasu, 2018-03-25:
The plot in my first post was "bi-static" RCS, but vineethajoy's expected plot was "mono-static" RCS. To get mono-static RCS from NEC2++'s bi-static RCS, 360 degree scattering pattern to be generated for each value of Phi and we have to extract the mono-static RCS value corresponding to the specific value of Phi amongst the 360 degree values.