tmolteno/necpp #57: Hardcoded atlas path in configure.ac doesn't support different linux OSes and non-standard atlas installation locations

asreimer, 2019-05-14:
On both Fedora 29 and CentOS 7, the configure script is unable to find atlas. It seems that the installation process assumes that atlas will be installed in a specific location, but that location is the same across different versions of linux.

On both, I have the standard `atlas` and `atlas-devel` packages installed. These packages install the atlas library into `/usr/lib64/atlas`, in contrast to the `/usr/lib/atlas-base/atlas` on Ubuntu and Debian.

I think the problem that I encountered is due to hardcoded paths in on lines 56 and 59 in the `configure.ac` file. The surrounding code uses `AC_CHECK_LIB`, but reading the autotools documentation, it seems that one should be using [`AC_SEARCH_LIBS` instead](https://autotools.io/autoconf/finding.html) to maximize portability.

In any case, I was able to get the configure script to work after changing the hardcoded: `/usr/lib/atlas-base/atlas` and `/usr/lib/atlas-base/atlas -llapack` to `/usr/lib64/atlas` and `/usr/lib64/atlas -lsatlas` on lines 56 and 59 of `configure.ac`. The latter change, `/usr/lib64/atlas -lsatlas`, which replaces the `-llapack` flag, is needed because atlas changed library names (see [here](https://lists.fedoraproject.org/archives/list/devel@lists.fedoraproject.org/thread/UTZ7ANWX2FCE3FLKTUYE3LM4U6QCS4LY/)). The `lapack` on line 58 also needs to be substituted for `satlas` and finally, one also needs to replace the `-llapack` on line 110 with `-lsatlas`.

Perhaps a simple solution is to provide both a `configure.ac.debian` and `configure.ac.redhat` and then as part of the installation instructions tell the use which to use?

At the very least, hopefully reporting this issue helps others install necpp on redhat-based linux distros. Cheers!

tmolteno, 2019-06-20:
Many thanks for this. It would be very nice to have a cross-platform way to manage the lapack install. A docker container might be the way to go.

ewheelerinc, 2021-08-31:
@asreimer, you might want -l**t**atlas because its multithreaded, but that depends on what you're doing I suppose.