tmolteno/necpp #58: Build fails on macOS due to linker error

dg1yfe, 2019-07-14:
Hi,

for some reason the linker fails when building on/for macOS (10.13.6):
```
[eligs@aquila ~/projects/necpp:master] make
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in src
/bin/sh ../libtool  --tag=CXX   --mode=link g++  -g -O2 -all-static  -o nec2++ nec2cpp.o XGetopt.o libnecpp.la -lm 
libtool: warning: complete static linking is impossible in this configuration
libtool: link: g++ -g -O2 -o nec2++ nec2cpp.o XGetopt.o -Wl,-bind_at_load  ./.libs/libnecpp.a -lstdc++ -lm
ld: warning: ignoring file ./.libs/libnecpp.a, file was built for archive which is not the architecture being linked (x86_64): ./.libs/libnecpp.a
```
Due to the lib not being used, a lot of 'undefined Symbol' errors follow:
```
Undefined symbols for architecture x86_64:
  "usage()", referenced from:
      nec_main(int, char**, nec_output_file&) in nec2cpp.o
  "secnds(double*)", referenced from:
      nec_main(int, char**, nec_output_file&) in nec2cpp.o
  "load_line(char*, __sFILE*)", referenced from:
      nec_main(int, char**, nec_output_file&) in nec2cpp.o
      readmn(__sFILE*, __sFILE*, char*, int*, int*, int*, int*, double*, double*, double*, double*, double*, double*) in nec2cpp.o
(and so on)
```

When I probe libnecpp.a with objdump, it tells me things are build for mach-o-x86-64:
```
[eligs@aquila ~/projects/necpp:master] objdump -a src/.libs/libnecpp.a        
In archive src/.libs/libnecpp.a:

libnecpp_la-c_evlcom.o:     file format mach-o-x86-64
rw-r--r-- 0/0 294608 Jan  1 01:00 1970 libnecpp_la-c_evlcom.o
(and so on...)
```
So ... when the lib is built for 64 Bit x86, why does the linker refuse to use it?
Would it be possible/of some use to link dynamically instead of static?
Since I'm not that familiar with autotools and linking against libraries, I'm kind of lost here and would appreciate any help to get this project to build.
Thanks.

dg1yfe, 2019-07-14:
Some digging in the web resolved this issue (still don't understand the reasons):
After disabling the binutils installed via homebrew ('brew unlink binutils'), the project builds.

tmolteno, 2019-07-22:
Thanks for looking into this. I'll leave this issue open, and close it after documenting in INSTALL.md

closed 2026-06-26 by tmolteno