tmolteno/necpp #62: cannot build with lapack support

nackstein, 2019-09-09:
I'm trying to configure with lapack support but I get an error. without lapack support I can build and install.

debugging into the problem I found this:
gcc -o conftest -g -O2 -L/usr/lib/lapack/ conftest.c -llapack
/usr/lib/gcc/i586-suse-linux/9/../../../../i586-suse-linux/bin/ld: /tmp/ccNWVBxh.o: in function `main':
/home/vortex/necpp-master/conftest.c:35: undefined reference to `clapack_zgetrf'
collect2: error: ld returned 1 exit status

readelf -Ws /usr/lib/lapack/liblapack.so.3.8.0 |grep zgetrf
  1376: 0016eb00  1163 FUNC    GLOBAL DEFAULT   13 zgetrf_
  1642: 00026e00  1403 FUNC    GLOBAL DEFAULT   13 zgetrf2_

for what I understand the configure create a conftest.c file and try to link against lapack calling clapack_zgetrf but this function is simply called zgetrf_ in the library.
I built lapack myself with rpmbuild on opensuse.

thanks for looking into this issue
Luigi

tmolteno, 2019-11-14:
Your system requires the lapack libraries in order to pass that test. These are available in the libatlas-base-dev package on Debian based systems. 

    sudo apt-get install libatlas-base-dev

should do the trick if you'd like to compile with lapack accelerated solvers.

ly2ss, 2020-05-23:
Hi,
Thank you for the software!
ubuntu 18.0.4lts with libatlas-base-dev installed. 

checking for clapack_zgetrf in -llapack... no
configure: error: in `/src/necpp':
configure: error: lapack library test failed (--without-lapack to disable)

It builds with without-lapack ok, but I am trying to model huge files (at least for me - 6000+ wires) and I would like to know how to speed things up.

ewheelerinc, 2021-08-31:
Hello all, we have the same issue on CentOS 7: 

- lapack-devel-3.4.2-8.el7.x86_64
- atlas-devel-3.10.1-12.el7.x86_64

```
checking for clapack_zgetrf in -llapack... no

configure:16784: checking for clapack_zgetrf in -llapack
configure:16817: gcc -o conftest -I/usr/include   conftest.c -llapack  -llapack -lblas   -L/usr/lib/gcc/x86_64-redhat-linux/4.8.5 -L/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib/gcc/x86_64-redhat-linux/4.8.5/../../.. -lgfortran -lm -lquadmath >&5
/tmp/ccEND97d.o: In function `main':
conftest.c:(.text+0xa): undefined reference to `clapack_zgetrf'

configure: error: in `/usr/src/upstream/nec2-cpp':
configure: error: lapack library test failed (--without-lapack to disable)

```
It looks like the function might be available in libsatlas or libtatlas, does that make sense?

```
# rpm -ql atlas-devel-3.10.1-12.el7.x86_64|xargs grep zgetrf
/usr/include/clapack.h:int clapack_zgetrf(const enum CBLAS_ORDER Order, const int M, const int N,
Binary file /usr/lib64/atlas/libsatlas.so matches
Binary file /usr/lib64/atlas/libtatlas.so matches

```

ewheelerinc, 2021-08-31:
Linking to this issue, sounds related: 
https://github.com/tmolteno/necpp/issues/57

ewheelerinc, 2021-09-08:
FYI, this link describes how to patch configure.ac to support OpenBLAS and ATLAS and (probably) fix this issue: https://stackoverflow.com/a/69095758/14055985

I was able to copy the configure.ac content and include PSFEX's .m4 macro files and build it with HAVE_ATLAS and HAVE_OPENBLAS defines, so I know it works for CentOS 7 and probably other distributions too. I've not tried patching necpp but perhaps it will help.

(In my case it built into [xnec2c](https://www.qsl.net/5b4az/pkg/nec2/xnec2c/doc/xnec2c.html) which is an NEC2 implementation independent of necpp, however that project will use dlopen() and dynamically bind based on what is available.)

ewheelerinc, 2021-12-08:
In case it is helpful, here is the .c file that xnec2c uses to support LAPACK, OpenBLAS and Intel MKL as shared objects to auto-detect and dynamically link as available:

https://github.com/KJ7LNW/xnec2c/blob/master/src/mathlib.c

SimplebutComplex, 2022-03-03:
This is still a bug with Ubuntu 20.04.  Having to hack the config.ac.  Has anyone convert this project to using cmake?