tmolteno/necpp #63: With large diameter segment intersection check is buggy

schlatterbeck, 2020-02-18:
I'm trying to model the boom of a log-periodic antenna.
The boom has a diameter of 15mm. I'm trying to model the boom piece-wise (because it needs to be connected to the elements, the element-feeding is done via the boom).

I have the following two (perfectly legal imo) geo wires:

    geo->wire(1, 5, 0.0,   0.0, 0.0, 0.0, 0.123, 0.0, 0.075, 1.0, 1.0);
    geo->wire(2, 5, 0.0, 0.123, 0.0, 0.0, 0.244, 0.0, 0.075, 1.0, 1.0);

for which the following error occurs:

GEOMETRY DATA ERROR -- FIRST SEGMENT MIDPOINT OF WIRE #2 (TAG ID #2) INTERSECTS WIRE #1 (TAG ID #1)

of course it does *NOT* intersect. The error goes away if I reduce the diameter of the boom wire but this is a hack not a solution.

I'll try attaching a compilable example (adapted from example/test_cpp.cpp)

Note that I'd like an option to turn this overlap-check off. There are perfectly legal cases where your overlap-check returns an error like modelling the phasing stubs of a hb9cv antenna, a picture is here:
https://www.changpuak.ch/electronics/HB9CV.php -- these stubs do not contribute to the radiation if they are near to the elements, in my experience this can be modeled fine with nec2c or xnec2c.

schlatterbeck, 2020-02-18:
I've not managed to attach a file, you can just modify example/test_cpp.cpp to include the two lines in the message above, this will reproduce the error.

tmolteno, 2020-05-20:
Thanks for the bug report. I'll have a look at the intersection code, as it may have a problem. From memory if the midpoint of any wire segment intersects with another wire, the interaction matrix will be invalid. This will make all the remaining output invalid. One way out of this would be to increase the length of each segment (moving the first segment midpoint of wire 2 away from wire 1). This can be done by reducing the segment count (although the segment length must be less than 0.1 wavelengths). Also the segment length must be greater than two wire radii (in the extended thin wire kernel). https://www.nec2.org/part_3/secii.html is very helpful on this.

Adding a switch to ignore geometry errors is indeed one way out of this. However if the features are not intended to have any electromagnetic effect on the radiation pattern, then I wonder if  it wouldn't be better to not include them in the nec2 geometry, rather than hoping the numerical errors they introduce don't affect the predictions of the properties of the antenna. In my experience, they do.

schlatterbeck, 2020-05-20:
On Wed, May 20, 2020 at 04:29:49AM -0700, Tim Molteno wrote:
> Thanks for the bug report. I'll have a look at the intersection code,
> as it may have a problem. From memory if the midpoint of any wire
> segment intersects with another wire, the interaction matrix [...]

If you decide to keep the check, it would be nice if this check could be
turned off. The nec manual also tells us that parallel wires (too near
to each other) *can* result in instabilities. I've experienced cases
where this works very good. Your current check does not allow these
cases. The nec manual (p.9) tells us "When wires are parallel and very
close together, the segments should be aligned to avoid incorrect
current perturbation from offset match point and segment junctions.
Although extensive tests have not been conducted, it is safe to specify
that wires should be several radii apart." So it is not explicitly
forbidden.

My current use-case is antenna optimization with a genetic algorithm
(I'm using my pgapy wrapper for the parallel genetic algorithm lib
pgapack). With this it can occur that the GA generates parameters that
might be rejected by your checks and I wouldn't want to code around
them.

Note that there are other sources of numerical instabilities in the NEC
code which you cannot guard against. I've managed to create a model that
had a negative SWV! (this would mean that the antenna produces energy)

> Adding a switch to ignore geometry errors is indeed one way out of
> this. However if the features are not intended to have any
> electromagnetic effect on the radiation pattern, then I wonder if  it [...]

I've seen cases where they don't.
And coding around the checks when you're optimizing parameters via
another tool is a pain.

Thanks for looking into this!

Ralf
-- 
Dr. Ralf Schlatterbeck                  Tel:   +43/2243/26465-16
Open Source Consulting                  www:   http://www.runtux.com
Reichergasse 131, A-3411 Weidling       email: office@runtux.com

schlatterbeck, 2021-06-01:
I've found another case where it would be desirable to remove the check: In NEC when modelling transmission lines (with the TL card or explicitly using the network NT card, only the latter allows to model lossy transmission lines) it is common practice to have two wires in parallel: Since when attaching a transmission line to a wire segment, if the transmission line should be terminated by an impedance, this cannot be done by loading the segment because the load is *in series* with the transmission line, not in parallel.
So it's common practice to have two wires between the same two points in the model. This is perfectly valid NEC code and is not possible with your checks.

I would advocate that your checks are optional -- it would be nice if they could be called _explicitly_ on a model, so that the modeller/programmer can decide to ignore (some of) the checks or not perform them at all.

Example NEC input that terminates a transmission line in a resistance using an additional parallel wire. Note that the TL card allows to terminate the transmission line, too. But the terminating impedance is specified as a complex admittance which will not vary with frequency. So the only way to terminate a transmission line in a capacitance or inductance is with a parallel second wire. In the example wires 1 and 3 are parallel (same endpoints), wire 3 is loaded with a resistance while wire 1 is the endpoint of the transmission line. This example will simulate a correctly terminated line (with a dummy load). Not particularly useful in practice but a good example why it should be possible to turn off the checks.

```
CM NT-Card with lambda/2
CM Loading segment 3 with 50 Ohm
CM lambda/2 is 42.827494
CE
GW 1 1 0 0     0 0.005 0     0 0.002
GW 3 1 0 0     0 0.005 0     0 0.002
GW 2 1 0 0.005 0 0.005 0.005 0 0.002
GE 0
TL 2 1 1 1 50 42.827494
EK 1
EX 0 2 1 0 1 0 0 0 0 0
LD 0 3 1 1 50 0 0
FR 0 3 0 0 3.49 0.01
RP 0 3 3 0 0 0 90 180 0 0
EN
```