tmolteno/necpp #74: Build Failure on Arch Linux

fisik-yum, 2022-12-13:
Hello there,
I'm trying to build this project to no avail. Using the steps outlined in #70, the build proceeds further but still fails with this error. I am using the steps from Install.md to build necpp.
This is not the AUR package. I've tried using build flags "-j4" and changing the C++ that gcc is to use.
Any help would be appreciated. Thanks.
```
In file included from c_geometry.h:30,
                 from c_geometry.cpp:19:
nec_wire.h: In constructor ‘nec_wire::nec_wire(const nec_3vector&, const nec_3vector&, nec_float, int)’:
nec_wire.h:31:19: error: implicitly-declared ‘constexpr nec_3vector::nec_3vector(const nec_3vector&)’ is deprecated [-Werror=deprecated-copy]
   31 |                 : x0(a), x1(b), radius(in_radius), _tag_id(id)
      |                   ^~~~~
In file included from c_geometry.h:20:
math_util.h:219:23: note: because ‘nec_3vector’ has user-provided ‘nec_3vector& nec_3vector::operator=(const nec_3vector&)’
  219 |   inline nec_3vector& operator=(const nec_3vector& copy) {
      |                       ^~~~~~~~
nec_wire.h:31:26: error: implicitly-declared ‘constexpr nec_3vector::nec_3vector(const nec_3vector&)’ is deprecated [-Werror=deprecated-copy]
   31 |                 : x0(a), x1(b), radius(in_radius), _tag_id(id)
      |                          ^~~~~
math_util.h:219:23: note: because ‘nec_3vector’ has user-provided ‘nec_3vector& nec_3vector::operator=(const nec_3vector&)’
  219 |   inline nec_3vector& operator=(const nec_3vector& copy) {
      |                       ^~~~~~~~
nec_wire.h: In member function ‘nec_3vector nec_wire::parametrize(nec_float) const’:
nec_wire.h:37:35: error: implicitly-declared ‘constexpr nec_3vector::nec_3vector(const nec_3vector&)’ is deprecated [-Werror=deprecated-copy]
   37 |                 nec_3vector ret = x0;
      |                                   ^~
math_util.h:219:23: note: because ‘nec_3vector’ has user-provided ‘nec_3vector& nec_3vector::operator=(const nec_3vector&)’
  219 |   inline nec_3vector& operator=(const nec_3vector& copy) {
      |                       ^~~~~~~~
nec_wire.h:39:24: error: implicitly-declared ‘constexpr nec_3vector::nec_3vector(const nec_3vector&)’ is deprecated [-Werror=deprecated-copy]
   39 |                 return ret;
      |                        ^~~
math_util.h:219:23: note: because ‘nec_3vector’ has user-provided ‘nec_3vector& nec_3vector::operator=(const nec_3vector&)’
  219 |   inline nec_3vector& operator=(const nec_3vector& copy) {
      |                       ^~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [Makefile:798: test_manager-c_geometry.o] Error 1
```

StefanBruens, 2023-02-15:
Likely fixed with https://github.com/tmolteno/necpp/commit/4d6be66d79e737620f796e5073f60d0940f4384e

Unfortunately, there is no release which includes this change ...

tmolteno, 2023-02-16:
I'm happy to do a release. if this helps. Are you building from the master branch?

    git pull
    make -f Makefile.git
    make -j 4

This works on debian linux.

fisik-yum, 2023-02-18:
Hi. Yes, if I remember correctly, I'm building from the master branch.
I would be thankful if you could do a release.

StefanBruens, 2023-02-18:
```
math_util.h:219:23: note: because ‘nec_3vector’ has user-provided ‘nec_3vector& nec_3vector::operator=(const nec_3vector&)’
  219 |   inline nec_3vector& operator=(const nec_3vector& copy) {
```

This ^ matches the line numbers of 1.6.1, but not the master branch for anything more recent than ~2016.

StefanBruens, 2023-02-19:
The error no longer appears when building the current master branch.