tmolteno/necpp #83: Is there anyone that have build nec2++ with newer Blas/intel and linux version (e.g centos 8)?

SV1TAN, 2024-12-27:
Hi,

I am using nec2++ for hf antennas design and i am very satisfy.
I would like to design antennas with reflectors in the microwave domain but the nec2++ is slow (6000+ segments...) in order to get accuracy.
I tried to build it in order to use all threads in my workstation like ytakeyasu issue #18 but I didn't manage to do it.
Maybe it has past a long time and there is no any compatibility with the newer versions of the Blas/intel libs or the linux.
So, I would to ask if someone have build the nec2++ with the newer versions of Blas/intel and linux version (e.g centos 8)?
If exist someone, I would like to please him to share with me the way how can I do this?

Thank you in advance,
Ragards,
SV1TAN (Spyros)

ytakeyasu, 2024-12-29:
What is the problem with compatibility, the Linux OS, the compiler or Intel MKL?　Using Intel MKL requires changes to the NEC2++ source code.　Have you changed the source code?

Yoshi Takeyasu
JA6XKQ

SV1TAN, 2024-12-29:
Thank you very much for your quick reply.
By doing all changes that you suggest in configure.ac and also in matrix_algebra.cpp. the Configuration passing but the compilation of the  matrix_algebra.cpp gives error for the lines that have added according to the #18.
The same error for openblas and for the mkl.
II am thinking maybe the definitions in matrix_algebra.cpp there are no more compatible with the new openblas mkl_2025 libraries...
I have also compile the openblas with the changes in the zgetrs.c but I get the same error.
I don't know... Also I am completely unrelated with programing (microwave engineer and radio amateur hobbyist).

Thanks
SV1TAN

ytakeyasu, 2024-12-29:
Could you please copy & paste the error message outputs by the compiler here?

I am not an expert on programming, but an expert on microwave engineering also. It would be helpful to describe the error message accurately, so that other programming experts can advise you on how to proceed.

SV1TAN, 2024-12-29:
A part of the compiler error...

matrix_algebra.cpp:418:61: error: ‘n’ was not declared in this scope
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                             ^
matrix_algebra.cpp:418:61: note: suggested alternative: ‘yn’
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                             ^
                                                             yn
matrix_algebra.cpp:418:77: error: ‘n’ was not declared in this scope
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                             ^
matrix_algebra.cpp:418:77: note: suggested alternative: ‘yn’
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                             ^
                                                                             yn
matrix_algebra.cpp:418:105: error: ‘a_in’ was not declared in this scope
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                                                         ^~~~
matrix_algebra.cpp:418:105: note: suggested alternative: ‘asin’
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                                                         ^~~~
                                                                                                         asin
matrix_algebra.cpp:418:131: error: ‘ndim’ was not declared in this scope
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                                                                                   ^~~~
matrix_algebra.cpp:418:131: note: suggested alternative: ‘fdim’
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                                                                                   ^~~~
                                                                                                                                   fdim
matrix_algebra.cpp:418:151: error: ‘ip’ was not declared in this scope
 fo = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
                                                                                                                                                 ^~

matrix_algebra.cpp:419:5: error: redefinition of ‘int info’
 int info = LAPACKE_zgetrs ((int) CblasColMajor, (char) CblasNoTrans, (lapack_int) n, (lapack_int) 1,  (const lapack_complex_double*) a.data(), (lapack_int) ndim, (const lapack_int*) ip.data(),  (lapack_complex_double*) b.data(), (lapack_int) n);
     ^~~~
matrix_algebra.cpp:418:5: note: ‘int info’ previously defined here
 int info = LAPACKE_zgetrf((int) CblasColMajor, (lapack_int) n, (lapack_int) n, (lapack_complex_double*) a_in.data(), (lapack_int) ndim, (lapack_int*) ip.data());
     ^~~~
matrix_algebra.cpp:419:83: error: ‘n’ was not declared in this scope
 int info = LAPACKE_zgetrs ((int) CblasColMajor, (char) CblasNoTrans, (lapack_int) n, (lapack_int) 1,  (const lapack_complex_double*) a.data(), (lapack_int) ndim, (const lapack_int*) ip.data(),  (lapack_complex_double*) b.data(), (lapack_int) n);
                                                                                   ^
matrix_algebra.cpp:419:83: note: suggested alternative: ‘yn’
 int info = LAPACKE_zgetrs ((int) CblasColMajor, (char) CblasNoTrans, (lapack_int) n, (lapack_int) 1,  (const lapack_complex_double*) a.data(), (lapack_int) ndim, (const lapack_int*) ip.data(),  (lapack_complex_double*) b.data(), (lapack_int) n);
                                                                                   ^
                                                                                   yn
.....
.....
.....

Thanks.

SV1TAN, 2024-12-29:
Can you please attach your configure.ac and the matrix_algebra.cpp file?
I would like to check if I define something wrong.

ytakeyasu, 2024-12-29:
I did not use autotools. I used Intel icpc as compiler and listed all the parameters required for linking on the command line. See Issue #18 for a description of those parameters. Which version of the source code did you use?　I used version 1.5.1 with modifications as reported in Issue #13, as there are still bugs in memory-allocation in versions newer than version 1.5.1. It is essential to use Blas/MKL to avoid memory-allocation problems for large scale simulations in the microwave domain. I will only show the source code once I know which version you are using.

SV1TAN, 2024-12-29:
I am using the latest code...
I will try with 1.5.1 version and I will let you know.

SV1TAN, 2024-12-29:
Making the appropriate changes as #18  I managed to compile the 1.5.3 version.
But when i run the nec2++ with a .nec file I get the output .out, but it works only 1 threads and also I get the error:
** On entry to ZGETRS parameter number  1 had an illegal value.
Do you know anything about this error?

Thanks.

ytakeyasu, 2024-12-29:
I didn't use the ver.1.5.3, but used the ver.1.5.1, though I don't know differences between them. No such errors have been experienced.

SV1TAN, 2024-12-29:
Finally I managed to compile the nec2pp with MKL.
But it works only with 1 thread...
Can the nec2pp use multi threads?
If yes, should I use any definition in the code in order to compile wit multithreading support?

Thanks.